awawa-dev/HyperSerialESP32 #4: esp32 hangs

Ilithia, 2022-10-13:
When screen is dimmed by UAC in windows, or when refresh rate or resolution changed, or when i alt+tab from exclusive fullscreen game, esp32 board turn leds off and won't turn on until i replug the board's usb and restart the program.

awawa-dev, 2022-10-13:
And restart the program alone doesn't resolve the problem? Provide full logs from the program when that happens. Probably it's grabber/capturing issue or just limitation.

Ilithia, 2022-10-17:
Restarting only the program doesn't do anything, i have to replug the board's usb and then restart the program

<details>
  <summary>Here's what happens when i run a program that requires elevated administrator privileges</summary>
  
```
2022-10-17T05:19:36.393Z [DX11_SYSTEM:AUTO] Lost DirectX capture context. Stopping.
2022-10-17T05:19:36.393Z [DX11_SYSTEM:AUTO] Stopped
2022-10-17T05:19:36.393Z [DX11_SYSTEM:AUTO] (DxGrabber.cpp:133) Uninit grabber: \\.\DISPLAY25|Radeon (TM) RX 480 Graphics
2022-10-17T05:19:37.418Z [MUXER0] Priority 245 is now inactive
2022-10-17T05:19:37.418Z [MUXER0] Set visible priority to 255
2022-10-17T05:19:37.418Z [HYPERHDR0] New priority[255], previous [245]
2022-10-17T05:19:37.418Z [HYPERHDR0] No source left -> switch LED-Device off
2022-10-17T05:19:37.418Z [SMOOTHING0] Clearing queued colors before: disabling
2022-10-17T05:19:37.418Z [SMOOTHING0] Smoothing queue is cleared
2022-10-17T05:19:37.426Z [LEDDEVICE_ADALIGHT] (LedDevice.cpp:136) Disable device
2022-10-17T05:19:37.427Z [LEDDEVICE_ADALIGHT] (LedDevice.cpp:336) Switch off
2022-10-17T05:19:37.427Z [LEDDEVICE_ADALIGHT] (LedDevice.cpp:294) Set LED strip to black/power off
2022-10-17T05:19:37.432Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:90) Flush was successful
2022-10-17T05:19:37.432Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:92) Close UART: COM3
2022-10-17T05:19:37.455Z [COMPONENTREG0] LED device: disabled
2022-10-17T05:19:37.479Z [DX11_SYSTEM:AUTO] (DxGrabber.cpp:170) init
2022-10-17T05:19:37.480Z [DX11_SYSTEM:AUTO] *************************************************************************************************
2022-10-17T05:19:37.480Z [DX11_SYSTEM:AUTO] Starting DX grabber. Selected: '\\.\DISPLAY25|Radeon (TM) RX 480 Graphics' max width: 512 (512) @ 30 fps
2022-10-17T05:19:37.480Z [DX11_SYSTEM:AUTO] *************************************************************************************************
2022-10-17T05:19:37.516Z [DX11_SYSTEM:AUTO] Selected D3D_DRIVER_TYPE_UNKNOWN
2022-10-17T05:19:37.517Z [DX11_SYSTEM:AUTO] Device initialized
2022-10-17T05:19:37.518Z [DX11_SYSTEM:AUTO] Started
2022-10-17T05:19:37.555Z [MUXER0] Priority 245 is now active
2022-10-17T05:19:37.555Z [MUXER0] Set visible priority to 245
2022-10-17T05:19:37.555Z [HYPERHDR0] New priority[245], previous [255]
2022-10-17T05:19:37.555Z [HYPERHDR0] New source available -> switch LED-Device on
2022-10-17T05:19:37.555Z [SMOOTHING0] Clearing queued colors before: enabling
2022-10-17T05:19:37.555Z [SMOOTHING0] Smoothing queue is cleared
2022-10-17T05:19:37.556Z [LEDDEVICE_ADALIGHT] (LedDevice.cpp:102) Enable device
2022-10-17T05:19:37.556Z [LEDDEVICE_ADALIGHT] Opening UART: COM3
2022-10-17T05:19:37.556Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:138) _rs232Port.open(QIODevice::ReadWrite): COM3, Baud rate [2000000]bps
2022-10-17T05:19:37.558Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:143) portName:          COM3
2022-10-17T05:19:37.558Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:144) systemLocation:    \\.\COM3
2022-10-17T05:19:37.558Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:145) description:       USB-Enhanced-SERIAL CH9102
2022-10-17T05:19:37.558Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:146) manufacturer:      wch.cn
2022-10-17T05:19:37.558Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:147) productIdentifier: 0x55d4
2022-10-17T05:19:37.558Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:148) vendorIdentifier:  0x1a86
2022-10-17T05:19:37.559Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:149) serialNumber:      5322005792
2022-10-17T05:19:37.559Z [LEDDEVICE_ADALIGHT] (LedDevice.cpp:310) Switch on
2022-10-17T05:19:37.560Z [LEDDEVICE_ADALIGHT] (LedDevice.cpp:386) Power On
2022-10-17T05:19:37.560Z [COMPONENTREG0] LED device: enabled
2022-10-17T05:19:37.560Z [SMOOTHING0] Using alternative smoothing input (0)
2022-10-17T05:19:37.576Z [SMOOTHING0] Using alternative smoothing procedure (0)
2022-10-17T05:19:42.303Z [DX11_SYSTEM:AUTO] (DxGrabber.cpp:486) AcquireNextFrame didn't return the frame. Just warning: the screen has not changed?
2022-10-17T05:19:42.337Z [DX11_SYSTEM:AUTO] (DxGrabber.cpp:486) AcquireNextFrame didn't return the frame. Just warning: the screen has not changed?
2022-10-17T05:19:42.435Z [DX11_SYSTEM:AUTO] (DxGrabber.cpp:486) AcquireNextFrame didn't return the frame. Just warning: the screen has not changed?
2022-10-17T05:19:42.501Z [DX11_SYSTEM:AUTO] (DxGrabber.cpp:486) AcquireNextFrame didn't return the frame. Just warning: the screen has not changed?
2022-10-17T05:19:42.567Z [DX11_SYSTEM:AUTO] (DxGrabber.cpp:486) AcquireNextFrame didn't return the frame. Just warning: the screen has not changed?
2022-10-17T05:21:00.021Z [PERFORMANCE] [INSTANCE0: FPS = 30.30, processed = 1818], [LED0: FPS = 50.00, send = 3000, processed = 3000]
2022-10-17T05:22:00.016Z [PERFORMANCE] [INSTANCE0: FPS = 30.30, processed = 1818], [LED0: FPS = 50.00, send = 3000, processed = 3000]
```
  
</details>
It says it's enabled, but the leds just stays black until i do the things i wrote before.

awawa-dev, 2022-10-17:
That's logs looks not quite good. The capturing warnings disappeared but it doesn't mean that the capturing is working: it's may just reach the number limit and you can capture just black frames so the LED strip will render black leds. For now nothing indicates the esp crash here but the problem with a capturing. The HyperHDR doesn't need UAC either for running: it may only need an access to the network because it's using web interface so your Windows firewall may ask you about permission once, at the first start but no more (in other circumstances something is wrong with your system or user's configuration). The logs are not full as I asked either but it's OK if you follow your own path to diagnose the problem on your system because I think I can't help here further since I can't reproduce the problem basing on the provided information.

closed 2022-10-17 by awawa-dev

Ilithia, 2022-10-18:
I didn't mean i run HyperHDR as admin, i run any other program that requires admin privileges.
 <details>
  <summary>here is the FULL log</summary>
  
  ```2022-10-18T02:07:24.747Z [MAIN] Database path: 'D:/Users/0/.hyperhdr/db/hyperhdr.db', readonlyMode = disabled
2022-10-18T02:07:24.747Z [MAIN] Starting HyperHdr - 18.0.0.0, (HEAD detached at 1c31205) (Awawa-e327d52/1c31205-1661084784), built: Aug 21 2022:14:05:03
2022-10-18T02:07:24.747Z [MAIN] Set user data path to 'D:/Users/0/.hyperhdr'
2022-10-18T02:07:24.749Z [DB] Database opened: D:/Users/0/.hyperhdr/db/hyperhdr.db
2022-10-18T02:07:24.848Z [SETTINGSMGR] Settings database initialized
2022-10-18T02:07:24.849Z [HYPERHDR] Sound device is disabled
2022-10-18T02:07:24.873Z [MEDIA_FOUNDATION:AUTO] Cropping image: width=0 height=0; crop: left=0 right=0 top=0 bottom=0 
2022-10-18T02:07:24.873Z [MEDIA_FOUNDATION:AUTO] Starting to enumerate video capture devices
2022-10-18T02:07:24.874Z [MEDIA_FOUNDATION:AUTO] (MFGrabber.cpp:413) Detected 0 devices
2022-10-18T02:07:24.874Z [MEDIA_FOUNDATION:AUTO] Restarting video grabber is now blocked due to reloading of the configuration
2022-10-18T02:07:24.874Z [MEDIA_FOUNDATION:AUTO] Cropping image: width=0 height=0; crop: left=0 right=0 top=0 bottom=0 
2022-10-18T02:07:24.874Z [Media Foundation:auto] (GrabberWrapper.cpp:75) CEC keycode. Start: 0, stop: 0
2022-10-18T02:07:24.874Z [MEDIA_FOUNDATION:AUTO] (Grabber.cpp:259) setBrightnessContrastSaturationHue nothing changed
2022-10-18T02:07:24.874Z [MEDIA_FOUNDATION:AUTO] (MFGrabber.cpp:175) setHdrToneMappingMode to: Disabled
2022-10-18T02:07:24.874Z [MEDIA_FOUNDATION:AUTO] (Grabber.cpp:168) setFpsSoftwareDecimation to: 1
2022-10-18T02:07:24.874Z [SIGNAL_OLD] (DetectionManual.cpp:85) Signal detection area set to: 0.250000,0.250000 x 0.750000,0.750000
2022-10-18T02:07:24.874Z [SIGNAL_OLD] (DetectionManual.cpp:75) Signal threshold set to: {12, 12, 12} and frames: 200
2022-10-18T02:07:24.874Z [SIGNAL_AUTO] (DetectionAutomatic.cpp:384) Automatic signal detection -> errorTolerance: 9, modelTolerance: 90, sleepTime: 5000, wakeTime: 1000
2022-10-18T02:07:24.874Z [MEDIA_FOUNDATION:AUTO] (Grabber.cpp:367) setDeviceVideoStandard preparing to restart video grabber. Old: '' new: 'auto'
2022-10-18T02:07:24.875Z [MEDIA_FOUNDATION:AUTO] Delayed restart of the grabber due to change of selected device
2022-10-18T02:07:24.875Z [MEDIA_FOUNDATION:AUTO] (Grabber.cpp:186) Force encoding to: NO_CHANGE (old: NO_CHANGE)
2022-10-18T02:07:24.875Z [MEDIA_FOUNDATION:AUTO] setQFrameDecimation is now: disabled
2022-10-18T02:07:24.875Z [Media Foundation:auto] (GrabberWrapper.cpp:545) Frame cache is: enabled
2022-10-18T02:07:24.875Z [DX11_SYSTEM:AUTO] Cropping image: width=0 height=0; crop: left=0 right=0 top=0 bottom=0 
2022-10-18T02:07:24.878Z [DX11_SYSTEM:AUTO] Restarting video grabber is now blocked due to reloading of the configuration
2022-10-18T02:07:24.879Z [DX11_SYSTEM:AUTO] (Grabber.cpp:145) Set new width: 512, height: 512 for capture
2022-10-18T02:07:24.879Z [DX11_SYSTEM:AUTO] (Grabber.cpp:322) setWidthHeight preparing to restarting video grabber 0
2022-10-18T02:07:24.879Z [DX11_SYSTEM:AUTO] Delayed restart of the grabber due to change of the capturing resolution
2022-10-18T02:07:24.879Z [DX11_SYSTEM:AUTO] Delayed restart of the grabber due to change of the framerate
2022-10-18T02:07:24.879Z [DX11_SYSTEM:AUTO] (DxGrabber.cpp:109) setHdrToneMappingMode to: Disabled
2022-10-18T02:07:24.879Z [SIGNAL_OLD] (DetectionManual.cpp:85) Signal detection area set to: 0.250000,0.250000 x 0.750000,0.750000
2022-10-18T02:07:24.879Z [SIGNAL_OLD] (DetectionManual.cpp:75) Signal threshold set to: {0, 0, 0} and frames: 50
2022-10-18T02:07:24.879Z [DX11_SYSTEM:AUTO] (Grabber.cpp:367) setDeviceVideoStandard preparing to restart video grabber. Old: '' new: 'auto'
2022-10-18T02:07:24.879Z [DX11_SYSTEM:AUTO] Delayed restart of the grabber due to change of selected device
2022-10-18T02:07:24.888Z [SETTINGSMGR] Settings database initialized
2022-10-18T02:07:24.889Z [BLACKBORDER] Set mode to: default
2022-10-18T02:07:24.889Z [COMPONENTREG0] Blackborder detector: enabled
2022-10-18T02:07:24.890Z [IMAGETOLED0] (ImageProcessor.cpp:163) Set LED mapping type to weighted
2022-10-18T02:07:24.890Z [IMAGETOLED0] (ImageProcessor.cpp:147) setSparseProcessing to 1
2022-10-18T02:07:24.890Z [RGB_TRANSFORM0] RGB transform classic_config: 1, saturationGain: 1.000000, luminanceGain: 1.000000, backlightThreshold: 0
2022-10-18T02:07:24.890Z [ADJUST_RED0] Set correction to 255
2022-10-18T02:07:24.890Z [ADJUST_BLUE0] Set correction to 255
2022-10-18T02:07:24.890Z [ADJUST_GREEN0] Set correction to 255
2022-10-18T02:07:24.956Z [JSONSERVER] (JsonServer.cpp:24) Created instance
2022-10-18T02:07:24.964Z [JSONSERVER] Started on port 19444
2022-10-18T02:07:25.005Z [MAIN] start systray
2022-10-18T02:07:24.892Z [HYPERHDR0] Led strip RGB order is: rgb
2022-10-18T02:07:24.893Z [DB] Database opened: D:/Users/0/.hyperhdr/db/hyperhdr.db
2022-10-18T02:07:24.966Z [WEBSERVER] Initialize Webserver
2022-10-18T02:07:24.966Z [WEBSERVER] Initialize Webserver
2022-10-18T02:07:24.966Z [FLATBUFSERVER] Started on port 19400
2022-10-18T02:07:24.967Z [FLATBUFSERVER] Started local domain socket server: 'hyperhdr-domain'
2022-10-18T02:07:24.971Z [SMOOTHING0] Clearing queued colors before: enabling
2022-10-18T02:07:24.971Z [SMOOTHING0] Smoothing queue is cleared
2022-10-18T02:07:24.971Z [COMPONENTREG0] Smoothing: enabled
[... cut ...]
2022-10-18T02:08:02.973Z [DX11_SYSTEM:AUTO] (DxGrabber.cpp:486) AcquireNextFrame didn't return the frame. Just warning: the screen has not changed?
2022-10-18T02:08:03.039Z [DX11_SYSTEM:AUTO] (DxGrabber.cpp:486) AcquireNextFrame didn't return the frame. Just warning: the screen has not changed?
2022-10-18T02:08:03.105Z [DX11_SYSTEM:AUTO] (DxGrabber.cpp:486) AcquireNextFrame didn't return the frame. Just warning: the screen has not changed?
2022-10-18T02:08:03.171Z [DX11_SYSTEM:AUTO] (DxGrabber.cpp:486) AcquireNextFrame didn't return the frame. Just warning: the screen has not changed?
  ```
  
</details>

awawa-dev, 2022-10-18:
DX grabber is reaching max warning limit (5) and still it's not clear if it's working properly or windows resources are not available for capturing. Try to run the animation  or set color from the effect tabs to test leds strip when that happens.

awawa-dev, 2022-10-19:
> Tried, and it's working normally, both static color and animations. 

So LED driver is working fine. But not DX capturing on your system in certain circumstances. You should notice it in the live preview from the start. But system grabber has it's own limitation and hardware grabber is recommended solution.

> I noticed there's a problem when i quit HyperHDR, the LEDs goes black 

that's expected.

> and won't turn on automatically on next HyperHDR launch until i reset the board by the button on it or replug it's usb and run the program after that

if there is no errors in the logs and you have **a video source** (or playing an effecting) then it's a hardware problem with your ESP32.