awawa-dev/HyperSerialESP32 #5: Compatibility with ESP32-S2

alucryd, 2022-11-07:
Hi, forgive the dumb question, I am completely new to ESP boards. Would the board below be compatible with HyperSerialESP32?
ESP32-S2 seems to have much better power efficiency, and the board's serial port advertises a maximum speed of 4.6Mbps.

https://www.kubii.fr/micro-controleurs/3651-carte-de-developpement-nodemcu-esp-12k-avec-wifi-3272496309937.html

awawa-dev, 2022-11-07:
Hi
S2 is not officially supported. You need to compile the sketch yourself and test it. You may experience performance and other exotic problem using newer Esp32 board manager unless they fixed them... esp32-S2 probably is not supported by older, stable versions. I don't have such ESP board so can't test it myself. As for a power efficiency HyperSerialESP32 doesnt use the Wifi radio so probably typical power usage is under 20mA instead ~300mA when the Wifi is ON. So I think this advantage of S2 is completely negligible when using HyperSerialESP32. I recommend standard ESP32 version (not S2,S3 etc.) equipped with CH9102x: 4Mb is confirmed to work for such boards.

alucryd, 2022-11-07:
Thanks for the quick answer. All the regular ESP32 boards I could find at my usual retailers were using the CP2102 you said we should avoid, and they didn't have the CH340G either. I'll seek a regular ESP32 with a CH9102x instead then, thanks for the suggestion. Might give an s2 or s3 a try when I have a working setup, could be a fun little project. Now to see if HyperHDR can talk to the ESP32 through a USB port of my webOS TV.

closed 2022-11-07 by alucryd

awawa-dev, 2022-11-08:
HyperSerialESP32 was just migrated to PlatformIO so compiling should be much easier.
[More here](https://github.com/awawa-dev/HyperSerialESP32/pull/6)
I testes also latest [Expressif32 5.2.0](https://github.com/platformio/platform-espressif32/releases/tag/v5.2.0) that provides Arduino 2.0.5 Unfortunately the bug hasn't been fixed yet and you will experience very degraded performance using Arduino 2.x for ESP32 ( [more here](https://github.com/awawa-dev/HyperHDR/discussions/267#discussioncomment-2784502) ). And without Arduino 2.x you won't be apply to compile the sketch for ESP32 S2.

alucryd, 2022-11-09:
Thank you for the follow up. I was able to get my hands on a FireBeetle from DFRobot (DFR0654) equipped with a CH340K chip, it's ESP32-E but the product page says it's software compatible with regular ESP32. Building using PlatformIO went without a hitch, as did flashing, I was able to add it to HyperHDR as an adalight device following the readme. Now I just have to wait for my LEDs to arrive to confirm everything is working. Thanks for the help and keep up the good work!

awawa-dev, 2022-11-09:
Great! You can also confirm if the serial communication works fine for your board at @2Mb speed using any serial port client for example putty: https://github.com/awawa-dev/HyperSerialEsp8266/discussions/20 Statistics should be updated if you run HyperHDR before and play some effects or use your video source as input.

alucryd, 2022-11-10:
Stats went from 0 frames to 19 frames after some time in HyperHDR looking at the live preview, serial seems to be working :)

alucryd, 2022-12-01:
Quick follow-up, I received an SK6812 3535 LED strip, however it would seem my board is not compatible after all. Connecting the data line to GPIO 2/D9 only produces random garbage akin to christmas lights. Also my TV's linux firmware doesn't come with a ch341 driver so I can't use it via serial, I had to switch to HyperSerialWLED.

WLED also produces garbage, even when I manually set a fixed color. I'm sure the GPIO pin is the right one because if I change it in WLED the christmas lights stop. I'll try to find another board.

awawa-dev, 2022-12-01:
As far I know (basing on users reports not my experience so can't guarantee)  CH9102x that comes with some ESP32 is supported by most webos. BTW there is a chance  for S2 support,  I must do more tests.

alucryd, 2022-12-01:
Thanks, I ordered a board from adafruit with a CH9102F chip, will see how that one goes.

Meanwhile I found an old ESP8266 NodeMCU board lying around, albeit with a CP2102 chip, but even with vanilla WLED I wasn't able to get the led strip to work properly, I'm beginning to wonder whether the strip could be faulty.

awawa-dev, 2022-12-01:
did you set a common ground between esp and the led strip? theoretically you probably missing also 3.3v to 5v level shifter.

alucryd, 2022-12-01:
I was reading one of your posts on the website which mentioned a common ground, and I realized I didn't have one. So that was a rookie mistake, my current setup is USB power for the board and an external power supply for the strip, connection the ground GPIO to the power supply's ground was the missing piece, thanks a lot! Now looking forward to receiving the adafruit board if it means I can use usb serial :)

alucryd, 2022-12-03:
The feather arrived faster than expected, the CH9102F can handle up to 4Mbps, and is supported by webos \o/

I had to change the data pin to something other than 2 though, I believe this is because GPIO 2 is on a special STEMMA QT connector, so I used 4 instead and it works like a charm.

This is the board if you want to document the findings somewhere: https://learn.adafruit.com/adafruit-esp32-feather-v2/pinouts

Thanks for the help and for the great piece of software!

awawa-dev, 2022-12-03:
Yes, pin2 seems to enable/disable STEMMA QT and should be not used in HyperSerialESP32 as an output Hmm...does this STEMMA QT have a 3.3v to 5v level shifter for neopixel leds? That would be great. As I understand pin2 needs to be set high and HyperSerialESP32 should use pin0.

alucryd, 2022-12-03:
They mention a 3.3v regulator, but no level shifter, I think you still need an external level shifter when using that connector.

awawa-dev, 2022-12-03:
BTW I've just finished testing ESP32 S2 mini... it works! That board is capable of 5Mb speed,

alucryd, 2022-12-04:
Cool! Thanks for the heads up, my next project will probably use one of those as they seem more readily available at my usual retailers.

incarvr6, 2023-12-23:
> BTW I've just finished testing ESP32 S2 mini... it works! That board is capable of 5Mb speed,

What is full steps to flash it from windows. It looks that esphome web flashing is not supported for S2.  A bit more than esptool.py information would be appreciated.

awawa-dev, 2023-12-23:
esptool.py is a Python application that is not maintained by us, but it is quite popular and there should be instructions available to find online. General tips: install Python first (when installing, remember to add it to your PATH), launch a Windows terminal and install esptool.py using Python's pip manager and it should work.

incarvr6, 2023-12-23:
> esptool.py is a Python application that is not maintained by us, but it is quite popular and there should be instructions available to find online. General tips: install Python first (when installing, remember to add it to your PATH), launch a Windows terminal aka 'command prompt' and install esptool.py using Python's pip manager and it should work.

After googling around and correctly preparing windows, flashing process was easy.
But now after flashing serial firmware I realised that pi zero2w have only one usb port, so looks like I will have to reflash to hyperspi.
I know it is off topic but I noticed on wiring examples that power for S2 mini is not taken from pi, is there any reason for that?

awawa-dev, 2023-12-24:
I always recommend using USB as a power source for ESP boards. I think I explained to you in another topic that getting 5V from the RPi GPIO can be tricky and may depend on the RPi model or even their version. Nevertheless, it can work if connected directly to the USB power connector, without any fuse or that crappy RPi PMIC along the way. However, with such power supply solution, you may bypass ESP protections (if the ESP board has them), which do not protect the GPIO.