awawa-dev/HyperSerialESP32 #12: Multi-Segment - How to?

dzirkler, 2022-12-16:
I'm excited to try out the mutli-segment feature, as I'm just about to replace my existing led strip with a 144/m strip in the next week.   I'm able to run a full 144 lights, so I know it's communicating with HyperHDR but I don't know how to check if it's using multi-segment. 

I've flashed my device with the latest build, but what I'm not sure of is if we need to wire in an additional GPIO for the second segment data?

The only note in the read me is this:
>Multi-segments can double your large sk6812/ws2812b setup refresh rate for free. All you need is to properly project & construct the LED strip and use HyperSerialESP32 v8.

But you haven't noted what "properly" means here, nor if there's any other configuration needed. 

Thanks for any additional notes or documentation!

awawa-dev, 2022-12-17:
Hi
I have limited access to internet now, but the proposed setup is:
- divide the long single into smaller 2 equal parts. So second index start in the HyperSerialESP32 firmware is the number of LEDs divided by 2.
- build your first segment traditional way e.g. clockwise, so it starts somewhere in middle of the bottom of frame/TV and end in the middle of the top of frame/TV
- start the second segment in the opposite direction to the first one e.g. counterclockwise (reversed option in the HyperSerialESP32 firmware configuration must be enabled). So it starts somewhere in the bottom of frame/TV and it ends in the middle of top of of frame TV/frame. Both segments should be connected at the top but only 5v and ground ( NOT the data line).
- the data line starts for both segments at the bottom
- configuration in HyperHDR does not change! It's should be configurators as one, single continues segment. All is done in HyperSerialESP32 firmware transparently and does not affect LED strip configuration in HyperHDR.

dzirkler, 2022-12-18:
* What data pin should be used for the second segment? 
* Do I need to do a build of the firmware myself with special settings?
* How does the firmware know to use the multisegment mode?

awawa-dev, 2022-12-19:
> * What data pin should be used for the second segment? 

E.g. gpio1 should be fine if not reserved by your esp32 board for its purpose.

> * Do I need to do a build of the firmware myself with special settings?

Yes, second segment start index and reversed options in the platformio.ini The options are explained there and how to build the firmware  online is explained in the wiki.

> * How does the firmware know to use the multisegment mode?

 As above.

dzirkler, 2022-12-19:
Thanks for this info, I'll give it a try this week, and if it all works out, I'll see if I can put some info into the readme with a PR to help others.