awawa-dev/HyperSerialESP32 #13: Can't run on webOS 7 LG OLED C2

nurikk-sa, 2022-12-20:
Hello! Thanks for a great project! :)
I'm trying to make it work on my rooted lg tv using esp32 dev kit with CH9102F usb2uart chip, which is supported natively by tv os.
But it doesn't work some how. In hyperhdr logs I can see that it successfully opens uart interface and manages to complite handshake, but after that nothing happens, no colour changes on led strip.  I'm using latest v19.0.0.0beta2 hypehdr and v8.0.0.0 hyperserial driver.
I've verified that this board and hyperhr works using my laptop running mac os, everything works as expected but when board is hocked to tv it doesn't work

https://github.com/webosbrew/hyperhdr-webos-loader/issues/9

```
2022-12-20T17:48:54.182Z [LEDDEVICE_ADALIGHT] Opening UART: ttyACM1
2022-12-20T17:48:54.182Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:174) _rs232Port.open(QIODevice::ReadWrite): ttyACM1, Baud rate [2000000]bps
2022-12-20T17:48:54.205Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:179) portName:          ttyACM1
2022-12-20T17:48:54.205Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:180) systemLocation:    /dev/ttyACM1
2022-12-20T17:48:54.205Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:181) description:       USB Single Serial
2022-12-20T17:48:54.205Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:182) manufacturer:      1a86
2022-12-20T17:48:54.205Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:183) productIdentifier: 0x55d4
2022-12-20T17:48:54.206Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:184) vendorIdentifier:  0x1a86
2022-12-20T17:48:54.206Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:185) serialNumber:      5539028341
2022-12-20T17:48:54.520Z [LEDDEVICE_ADALIGHT] DETECTED DEVICE USING HYPERSERIALESP8266/HYPERSERIALESP32 FIRMWARE (Welcome! Awa driver 8. SPI APA102 compatible type (BGR).) at 47 msec
2022-12-20T17:48:54.520Z [LEDDEVICE_ADALIGHT] (LedDevice.cpp:329) Switch on
2022-12-20T17:48:54.521Z [LEDDEVICE_ADALIGHT] (LedDevice.cpp:405) Power On
2022-12-20T17:48:54.521Z [COMPONENTREG0] LED device: enabled
2022-12-20T17:48:57.798Z [WEBSOCKET] (JsonAPI.cpp:1250) log streaming activated for client ::ffff:192.168.4.23
2022-12-20T17:49:20.949Z [WEBSOCKET] (JsonAPI.cpp:1259) log streaming deactivated for client  ::ffff:192.168.4.23
2022-12-20T17:49:20.950Z [WEBSOCKET] (JsonAPI.cpp:1250) log streaming activated for client ::ffff:192.168.4.23
```

```
root@LGwebOSTV:/tmp/var/log# uname -a
Linux LGwebOSTV 5.4.96-266.mlt4tv.1 #1 SMP PREEMPT Tue Nov 8 12:52:11 UTC 2022 aarch64 GNU/Linux
```

```
System info

None
- Database:        read/write

HyperHDR Server OS: 
- Distribution:   webOS TV 7.2.0
- Architecture:   arm64
- Kernel:         linux (5.4.96-266.mlt4tv.1 (WS: 32))
- Qt Version:     5.15.8
```

awawa-dev, 2022-12-20:
Hi
Since webos it is not officially supported in any of my projects because I don't have proper testing setup, I can only guess. What are you HyperSerialESP32 statistics taken from HyperHDR after you play some effects for at least few seconds and then disable the LED device? Did you try to reset the TV and/or disable the quick start?  Last thing you should try is to disable ESP handshake if for some reason webos doesn't like when serial device is reset itself (but HyperHDR log doesn't confirm it, looks fine). You could manually reset esp32 board with the button when LED is disabled in HyperHDR and enable it again next (with ESP handshake still disabled).

nurikk-sa, 2022-12-21:
> Hi Since webos it is not officially supported in any of my projects because I don't have proper testing setup, I can only guess. What are you HyperSerialESP32 statistics taken from HyperHDR after you play some effects for at least few seconds and then disable the LED device? Did you try to reset the TV and/or disable the quick start? Last thing you should try is to disable ESP handshake if for some reason webos doesn't like when serial device resets itself (but HyperHDR log doesn't confirm it, looks fine). You could manually reset esp32 board with the button when LED is disabled in HyperHDR and enable it again next (with ESP handshake still disabled).

Resetting (RST) esp32 definitely helps, it starts working only after reset. Enabling/disabling Led device doesn't help. I wonder where's the issue. 

```
2022-12-21T12:40:57.879Z [WEBSOCKET] (JsonAPI.cpp:1250) log streaming activated for client ::ffff:192.168.4.23
2022-12-21T12:41:05.873Z [WEBSOCKET] (JsonAPI.cpp:1630) message: [{"command":"leddevice","ledDeviceType":"adalight","subcommand":"discover","tan":645}]
2022-12-21T12:41:05.903Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:340) Serial devices discovered: [{"devices":[{"name":"Auto","value":"auto"},{"name":" (/dev/ttyAMA0)","value":"ttyAMA0"},{"name":" (/dev/ttyAMA1)","value":"ttyAMA1"},{"name":" (/dev/ttyAMA2)","value":"ttyAMA2"},{"name":"USB Single Serial (/dev/ttyACM0)","value":"ttyACM0"}],"ledDeviceType":"adalight"}]
2022-12-21T12:41:05.903Z [WEBSOCKET] (JsonAPI.cpp:1654) response: [{"devices":[{"name":"Auto","value":"auto"},{"name":" (/dev/ttyAMA0)","value":"ttyAMA0"},{"name":" (/dev/ttyAMA1)","value":"ttyAMA1"},{"name":" (/dev/ttyAMA2)","value":"ttyAMA2"},{"name":"USB Single Serial (/dev/ttyACM0)","value":"ttyACM0"}],"ledDeviceType":"adalight"}]
2022-12-21T12:41:05.913Z [WEBSOCKET] (JsonAPI.cpp:1259) log streaming deactivated for client  ::ffff:192.168.4.23
2022-12-21T12:41:11.850Z [SMOOTHING0] Clearing queued colors before: disabling
2022-12-21T12:41:11.850Z [SMOOTHING0] Smoothing queue is cleared
2022-12-21T12:41:11.850Z [LEDDEVICE_ADALIGHT] (LedDevice.cpp:138) Disable device
2022-12-21T12:41:11.850Z [LEDDEVICE_ADALIGHT] (LedDevice.cpp:355) Switch off
2022-12-21T12:41:11.850Z [LEDDEVICE_ADALIGHT] (LedDevice.cpp:313) Set LED strip to black/power off
2022-12-21T12:41:11.863Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:123) Flush was successful
2022-12-21T12:41:11.864Z [COMPONENTREG0] LED device: disabled
2022-12-21T12:41:11.865Z [PERFORMANCE] [INSTANCE0: FPS = 49.98, processed = 2999]
2022-12-21T12:41:13.151Z [WEBSOCKET] (JsonAPI.cpp:1250) log streaming activated for client ::ffff:192.168.4.23
2022-12-21T12:41:14.872Z [LEDDEVICE_ADALIGHT] Received: '
HyperHDR frames: 23 (FPS), receiv.: 25, good: 23, incompl.: 2, mem: 3316, heap: 359820
HyperHDR frames: 23 (FPS), receiv.: 25, good: 23, incompl.: 2, mem: 3316, heap: 359820' (176)
2022-12-21T12:41:14.872Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:106) Close UART: ttyACM0
2022-12-21T12:41:16.609Z [PERFORMANCE] [INSTANCE0: FPS = 26.32, processed = 1581]
2022-12-21T12:41:35.131Z [EFFECT0(Atomic...)] The effect quits with priority: 1
2022-12-21T12:41:35.131Z [MUXER0] Removed source priority 1
2022-12-21T12:41:35.131Z [MUXER0] Set visible priority to 150
2022-12-21T12:41:35.131Z [HYPERHDR0] New priority[150], previous [1]
2022-12-21T12:41:35.132Z [EFFECTENGINE0] Effect 'Atomic swirl' has finished.
2022-12-21T12:41:35.132Z [EFFECT0(Atomic...)] Deleting effect named: 'Atomic swirl'
2022-12-21T12:41:35.132Z [EFFECT0(Atomic...)] Effect named: 'Atomic swirl' is deleted
2022-12-21T12:41:35.133Z [IMAGETOLED0] Total index number is: 13558 (memory: 13558). User sparse processing is: disabled, image size: 320 x 180, area number: 231
2022-12-21T12:41:47.615Z [SMOOTHING0] Clearing queued colors before: enabling
2022-12-21T12:41:47.615Z [SMOOTHING0] Smoothing queue is cleared
2022-12-21T12:41:47.615Z [LEDDEVICE_ADALIGHT] (LedDevice.cpp:102) Enable device
2022-12-21T12:41:47.615Z [LEDDEVICE_ADALIGHT] Opening UART: ttyACM0
2022-12-21T12:41:47.616Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:183) _rs232Port.open(QIODevice::ReadWrite): ttyACM0, Baud rate [2000000]bps
2022-12-21T12:41:47.620Z [WEBSOCKET] (JsonAPI.cpp:1259) log streaming deactivated for client  ::ffff:192.168.4.23
2022-12-21T12:41:47.641Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:188) portName:          ttyACM0
2022-12-21T12:41:47.641Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:189) systemLocation:    /dev/ttyACM0
2022-12-21T12:41:47.642Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:190) description:       USB Single Serial
2022-12-21T12:41:47.642Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:191) manufacturer:      1a86
2022-12-21T12:41:47.642Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:192) productIdentifier: 0x55d4
2022-12-21T12:41:47.642Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:193) vendorIdentifier:  0x1a86
2022-12-21T12:41:47.642Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:194) serialNumber:      5539028341
2022-12-21T12:41:47.907Z [LEDDEVICE_ADALIGHT] Could not detect HyperSerialEsp8266/HyperSerialESP32 device
2022-12-21T12:41:47.907Z [LEDDEVICE_ADALIGHT] (LedDevice.cpp:329) Switch on
2022-12-21T12:41:47.907Z [LEDDEVICE_ADALIGHT] (LedDevice.cpp:405) Power On
2022-12-21T12:41:47.907Z [COMPONENTREG0] LED device: enabled
2022-12-21T12:41:47.908Z [SMOOTHING0] Clearing queued colors before: enabling. Smoothing configuration changed: restarting timer.
2022-12-21T12:41:47.908Z [SMOOTHING0] Smoothing queue is cleared
[... cut ...]
2022-12-21T12:41:47.928Z [SMOOTHING0] Using alternative smoothing procedure (0)
2022-12-21T12:41:50.667Z [MUXER0] Register new input 'Web Configuration@::ffff:192.168.4.23/COLOR' with priority 1 as inactive
2022-12-21T12:41:50.668Z [MUXER0] (PriorityMuxer.cpp:219) Priority 1 is now active
2022-12-21T12:41:50.668Z [MUXER0] Set visible priority to 1
2022-12-21T12:41:50.668Z [HYPERHDR0] New priority[1], previous [150]
2022-12-21T12:41:58.418Z [WEBSOCKET] (JsonAPI.cpp:1250) log streaming activated for client ::ffff:192.168.4.23
```

awawa-dev, 2022-12-21:
> Resetting (RST) esp32 definitely helps, it starts working only after reset. Enabling/disabling Led device doesn't help. I wonder where's the issue.
>
> ```
> 2022-12-21T12:40:57.879Z [WEBSOCKET] [...]

HyperHDR ESP handshake also resets the board using RST/dtr at the begining and set proper state. Don't know what is causing the issue.

nurikk-sa, 2022-12-21:
>HyperHDR ESP handshake also resets the board using RST/dtr at the begining and set proper state. Don't know what is causing the issue.

can reset workflow fail if in the beginning there's some messages from bootloader?

nurikk, 2022-12-21:
I’ve managed to reproduce same issue on my laptop running mac os.
Sometimes esp handshake doesn’t succeed

awawa-dev, 2022-12-21:
> can reset workflow fail if in the beginning there's some messages from bootloader?

In your logs the reset succeeded, there is an initial message from HyperSerialESP32 when the board with uploaded firmware starts again. Again, check the statistics when it doesn't work as it's described in ESP handshake help: it's important to see if there are any or if the communication is broken. It's a wide shot but if you have SPI led strip connected to GPIO0 change it to other GPIO (e.g. 4) GPIO0 could interfere in the boot processes triggering the bootloader mode while restarting, but your HyperHDR logs doesn't confirm such scenario and it's depends on your electrical scheme how it's connected.

nurikk, 2022-12-22:
> > can reset workflow fail if in the beginning there's some messages from bootloader?
> 
> In your logs the reset succeeded, there is an initial message from HyperSerialESP32 when the board with uploaded firmware starts again. Again, check the statistics when it doesn't work as it's described in ESP handshake help: it's important to see if there are any or if the communication is broken. It's a wide shot but if you have SPI led strip connected to GPIO0 change it to other GPIO (e.g. 4) GPIO0 could interfere in the boot process triggering the bootloader mode while restarting, but your HyperHDR logs doesn't confirm such scenario and it's depends on your electrical scheme how it's connected.

I have quite simple wiring to gpio 4/2.  It’s definitely esp initialisation issues. I’ve ordered couple of other boards just in case if current issue be is dodge

awawa-dev, 2022-12-22:
> I have quite simple wiring to gpio 4/2.  It’s definitely esp initialisation issues. I’ve ordered couple of other boards just in case if current issue be is dodge 

So in both cases the issue with the board affects SPI LEDs but it's handled by Makuna neopixel library (older version because we can't use latest Arduino ESP32 yet due to some performance problem with the serial port in 2.x) and beyond possible diagnostic in this project: unless somebody has other ideas. Bootloader mode can be disabled putting high (3.3v) on GPIO0 but verify it first because currently I have only mobile access to the internet.

nurikk-sa, 2022-12-22:
> because currently I have only mobile access to the internet.

I checked reset sequence, and during `intializeEsp` I only see 1us pulse signal on EN pin. I'll try to fiddle with reset sequence and update later

[image]

nurikk-sa, 2022-12-22:
I hooked up esp to my logic analyser and clearly see there's issue with esp firmware startup.  After reset pulse I only can see messages from bootloader. But not `\r\nWelcome!\r\nAwa driver 8.z` from firmware.

[image]

awawa-dev, 2022-12-22:
Before that initial message completely nothing happens for SPI (for RGBW there is setup of the calibration), even neopixel is initialize later. Try to remove serial.flush in the setup sequence but it does make no sense since sk6812 is not affected. To "wake up" the serial communication after the reset you may need to set proper DTR/RTS but again: the same sequence works for sk6812. Test without LED strip connected to eliminate electrical provlem

awawa-dev, 2022-12-22:
BTW esp32 boot is using 115200 if I remember correctly. The HyperSerialESP32 switches to 2Mb: you may need to increase the resolution for the logic analizer. Otherwise it may be seen as single pulse.

nurikk-sa, 2022-12-22:
> BTW esp32 boot is using 115200 if I remember correctly. Then HyperSerialESP32 switches to 2Mb: you may need to increase the resolution for the logic analizer. Otherwise it may be seen as a single pulse.

Yep, I have 2 separate decoders enabled for same input. 

I've found another board with CH9102 chip, it has the same issue

nurikk-sa, 2022-12-22:
And what's common with this boards (the're different manufacturers and pcb layouts), that I can't flash them without manually grounding GPIO0. I also found a github issue with similar problem https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/139

nurikk-sa, 2022-12-22:
Okay, just in case if someone else is looking: 
Hellobox from discord also compiled drivers uart drivers for webos 7

`USB serial drivers (ch341, cp210x)  for webOS 7/22 (kernel 5.4.96, 64-bit/AArch64, built against LG kernel source).`
[uart drivers.zip](https://github.com/awawa-dev/HyperSerialESP32/files/10290548/uart.drivers.zip)

https://discord.com/channels/407937994037919756/413498235307425802/1054412614396027011

awawa-dev, 2022-12-22:
> And what's common with this boards (the're different manufacturers and pcb layouts), that I can't flash them without manually grounding GPIO0. I also found a github issue with similar problem https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/139

With my board mh-et live I had to ground GPIO0 for the first time, to upload firmware with a bootloader. It came without it. Later when you have a bootloader you can upload/upgrade the firmware without using GPIO0, py-esptool can handle it.

nurikk-sa, 2022-12-22:
with cp210x driver and chip it's still same problem. Firmware doesn't start after rst pulse, only if I pre RST button manually 

```
2022-12-22T22:18:30.785Z [LEDDEVICE_ADALIGHT] Opening UART: ttyUSB0
2022-12-22T22:18:30.785Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:183) _rs232Port.open(QIODevice::ReadWrite): ttyUSB0, Baud rate [2000000]bps
2022-12-22T22:18:30.810Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:188) portName:          ttyUSB0
2022-12-22T22:18:30.810Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:189) systemLocation:    /dev/ttyUSB0
2022-12-22T22:18:30.811Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:190) description:       CP2104 USB to UART Bridge Controller
2022-12-22T22:18:30.811Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:191) manufacturer:      Silicon Labs
2022-12-22T22:18:30.811Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:192) productIdentifier: 0xea60
2022-12-22T22:18:30.811Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:193) vendorIdentifier:  0x10c4
2022-12-22T22:18:30.811Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:194) serialNumber:      01DF0242
2022-12-22T22:18:31.081Z [LEDDEVICE_ADALIGHT] Could not detect HyperSerialEsp8266/HyperSerialESP32 device
2022-12-22T22:18:31.082Z [LEDDEVICE_ADALIGHT] (LedDevice.cpp:329) Switch on
2022-12-22T22:18:31.082Z [LEDDEVICE_ADALIGHT] (LedDevice.cpp:405) Power On
2022-12-22T22:18:31.082Z [COMPONENTREG0] LED device: enabled
2022-12-22T22:18:31.082Z [SMOOTHING0] Using alternative smoothing input (0)
2022-12-22T22:18:31.082Z [SMOOTHING0] Using alternative smoothing procedure (0)
2022-12-22T22:18:44.527Z [WEBSOCKET] (JsonAPI.cpp:1250) log streaming activated for client ::ffff:192.168.4.23
2022-12-22T22:20:44.769Z [SMOOTHING0] Clearing queued colors before: disabling
2022-12-22T22:20:44.769Z [SMOOTHING0] Smoothing queue is cleared
2022-12-22T22:20:44.769Z [LEDDEVICE_ADALIGHT] (LedDevice.cpp:138) Disable device
2022-12-22T22:20:44.769Z [LEDDEVICE_ADALIGHT] (LedDevice.cpp:355) Switch off
2022-12-22T22:20:44.769Z [LEDDEVICE_ADALIGHT] (LedDevice.cpp:313) Set LED strip to black/power off
2022-12-22T22:20:44.774Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:123) Flush was successful
2022-12-22T22:20:44.776Z [COMPONENTREG0] LED device: disabled
2022-12-22T22:20:44.777Z [WEBSOCKET] (JsonAPI.cpp:1259) log streaming deactivated for client  ::ffff:192.168.4.23
2022-12-22T22:20:46.435Z [WEBSOCKET] (JsonAPI.cpp:1250) log streaming activated for client ::ffff:192.168.4.23
2022-12-22T22:20:47.786Z [LEDDEVICE_ADALIGHT] Received: '
HyperHDR frames: 22 (FPS), receiv.: 22, good: 22, incompl.: 0, mem: 3316, heap: 359024
HyperHDR frames: 22 (FPS), receiv.: 22, good: 22, incompl.: 0, mem: 3316, heap: 359024' (176)
2022-12-22T22:20:47.786Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:106) Close UART: ttyUSB0
```

nurikk-sa, 2022-12-23:
okay, for the future google people:

https://discord.com/channels/407937994037919756/974942819707924511/1023685372905009233

```
asturel — 09/25/2022 9:00 PM
Some OS and/or drivers may activate RTS and or DTR automatically when opening the serial port (true only for some serial terminal programs, not esptool.py), pulling them low together and holding the ESP in reset. If RTS is wired directly to EN then RTS/CTS “hardware flow control” needs to be disabled in the serial program to avoid this. An additional circuitry is implemented in order to avoid this problem - if both RTS and DTR are asserted together, this doesn’t reset the chip. The schematic shows this specific circuit with two transistors and its truth table.
https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html
If this circuitry is implemented (all Espressif boards have it), adding a capacitor between the EN pin and GND (in the 1uF-10uF range) is necessary for the reset circuitry to work reliably. This is shown in the ESP32 Module section of the schematic.
```

```
asturel — Today at 12:36 AM
afaik nobody had properly working ch9102 on webos 😄 probably the usb cdc acm driver sends some RST / keeps esp in bootloop/stuck in /boot/ state 😄
```

10uF cap trick worked for me with CP2104 chip

nurikk-sa, 2022-12-23:
I've soldered electrolyte cap between EN/GND. This allowed esp to start from tv off state. The only problem left is that if I manually disable LED device from Remote Control page it doesn't come back to life.

nurikk, 2022-12-23:
Using longer reset pulse (150ms) solved issue with esp being in half booted state after reset.

awawa-dev, 2022-12-24:
@nurikk-sa could you check if above PR solves also your problem?

nurikk, 2022-12-24:
> @nurikk-sa could you check if above PR solves also your problem?

It’s the same person as me. Sorry messing up with personal/work github accounts from different browsers

awawa-dev, 2022-12-24:
Ah, OK then :) So I think this issue can be closed if it works for you and it's specific board/webos problem? I test PR later with my ESP8266/ESP32 boards collection but I think that increasing timing slightly should not be a problem (even if so, when could detect the board using productid/manufactureid and apply longer pulse, I also consider such solution. Could you post your esp32 IDs because for now I want to just compare it with mine boards?

awawa-dev, 2022-12-24:
Hmm... esptool from espressif is using even longer sleep: https://github.com/espressif/esptool/blob/65f861420515217f7b5d209fce288808dc5b1943/esptool/targets/esp32s3.py#L207

nurikk, 2022-12-24:
I doubt that even 0.5s reset pulse would be noticed by anyone. Since it’s just during board initialisation.

nurikk-sa, 2022-12-24:
Board1: with ch9102x chip

```
2022-12-24T17:58:50.077Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:188) portName:          cu.wchusbserial52870176471
2022-12-24T17:58:50.077Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:189) systemLocation:    /dev/cu.wchusbserial52870176471
2022-12-24T17:58:50.077Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:190) description:       USB Single Serial
2022-12-24T17:58:50.077Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:191) manufacturer:      
2022-12-24T17:58:50.077Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:192) productIdentifier: 0x55d4
2022-12-24T17:58:50.077Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:193) vendorIdentifier:  0x1a86
2022-12-24T17:58:50.077Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:194) serialNumber:      5287017647
```
[image]
[image]

nurikk-sa, 2022-12-24:
Board2: with 
```
2022-12-24T18:02:55.980Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:183) _rs232Port.open(QIODevice::ReadWrite): cu.wchusbserial55390283411, Baud rate [2000000]bps
2022-12-24T18:02:55.986Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:188) portName:          cu.wchusbserial55390283411
2022-12-24T18:02:55.987Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:189) systemLocation:    /dev/cu.wchusbserial55390283411
2022-12-24T18:02:55.988Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:190) description:       USB Single Serial
2022-12-24T18:02:55.988Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:191) manufacturer:      
2022-12-24T18:02:55.988Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:192) productIdentifier: 0x55d4
2022-12-24T18:02:55.988Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:193) vendorIdentifier:  0x1a86
2022-12-24T18:02:55.988Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:194) serialNumber:      5539028341
```
[image]
[image]

nurikk-sa, 2022-12-24:
board3: with cp1204, looks the same as board1

```
2022-12-24T16:59:02.751Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:188) portName:          ttyUSB0
2022-12-24T16:59:02.752Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:189) systemLocation:    /dev/ttyUSB0
2022-12-24T16:59:02.752Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:190) description:       CP2104 USB to UART Bridge Controller
2022-12-24T16:59:02.752Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:191) manufacturer:      Silicon Labs
2022-12-24T16:59:02.752Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:192) productIdentifier: 0xea60
2022-12-24T16:59:02.752Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:193) vendorIdentifier:  0x10c4
2022-12-24T16:59:02.752Z [LEDDEVICE_ADALIGHT] (ProviderRs232.cpp:194) serialNumber:      01DF0242
```

nurikk-sa, 2022-12-24:
all board had reset issue with 100ms pulse and works fine with 150ms pulse

awawa-dev, 2022-12-24:
Ok, thanks. The first looks like mine. It's not directly related (100 or 150ms doesnt make any difference) but with `sleep` during initializing there is one issue: the signal from the main thread is blocking till the LED startup/disable is finished so HyperHDR (all components running on the main thread) are paused then.

awawa-dev, 2022-12-24:
Tested two Esp8266 (mini and mini pro) and one generic ESP32 with ch340g: they also work fine with your PR. I have one issue with resetting ESP32 S2 Lolin Mini (refuses to reset itself) so maybe I'll experiment with that pulse time in future. PR can be merged now. Thank you.

closed 2022-12-24 by awawa-dev

daagaak, 2023-01-14:
Any thoughts as to why Lolin Mini's aren't resetting properly? I think I'm running into this on my LG CX + Mini S2 setup. Changing the setup of the HyperLED output frequently logs up the RGB output until I sleep the TV and wake it up again.