awawa-dev/HyperSerialESP32 #16: ESP-WROOM-32 boot loop issue

bruvv, 2023-01-07:
I had an issue with the pre compiled bin file. (specifically this one: https://github.com/awawa-dev/HyperSerialESP32/releases/download/v8.0.0.0/firmware_esp32_SK6812_RGBW_COLD.bin)

When compiling it myself it worked fine, I did change the platform version to the latest 5.3.0 (https://github.com/awawa-dev/HyperSerialESP32/blob/master/platformio.ini#L32) so far it has been absolutely incredible! Is there a reason you build it using version 3.5.0 or is it a typo?

awawa-dev, 2023-01-07:
Hi
No, fixing Arduino-ESP32 to version 3.5.0 was done on purpose because there was a bug introduced in Arduino-ESP32 2.0 that prevented HyperSerialESP32 from working correctly. https://github.com/espressif/arduino-esp32/issues/7505 But I see they've just released version 2.0.6 https://github.com/espressif/arduino-esp32/releases/tag/2.0.6 that includes a fix for it, so I will update platform version in HyperSerialESP32 as well.

bruvv, 2023-01-07:
Thanks! I will see if I can port it to esphome so people can use the easy to use OTA. Will close this for now.

closed 2023-01-07 by bruvv