awawa-dev/HyperSerialESP32 #30: can not get multi-segment sk6812 rgbw cold to compile and upload to lolin esp32-s2 mini

monpiere, 2023-03-04:
I have been trying all night to properly edit the platformio file to make the firmware for my esp32-s2 mini.  I'm attempting to run a multi segment setup with an sk6812 rgbw cold white led strip, lolin esp32-s2 mini, 74AHCT125 level shifter, Rpi 4, hyperserialesp32, and HyperHDR 19.0. I have windows 11 desktops to use to get all the programming done. I know that you are busy, but if you could please just hold my hand and walk me through this, I would appreciate it greatly!!!! The wiring isn't an issue, I've got that I'm fairly certain, and I'm pretty sure that I can get it taken care of after it's all loaded up. I had an old setup that was just the rpi4 and a ws2812b strip for a while, but chose to upgrade it due to instability.  Thank you so much for your time!!

Jason

awawa-dev, 2023-03-04:
Hi
For sure that part is wrong: `-DSERIALCOM_SPEED=5000000 -DATA_PIN = 2`
DATA_PIN is missing -D prefix and there are forbidden spaces around `...PIN = 2`
I'm not sure it wouldn't be easier to remove/re-fork and apply the changes to the sections you need, but it's your decision.

monpiere, 2023-03-04:
Okay great! That's exactly what I'll do. I just didn't know. I'm sorry to sound so ignorant, but I am exactly that, ignorant.

awawa-dev, 2023-03-04:
It's OK. This project uses a rather unusual solution to deliver custom builds of firmware, so it's easy to make a mistake when configuring it. Thank you for your support :)

This line looks OK so can be used again in esp32-s2 sk6812 cold white section (`[env:s2_mini_SK6812_RGBW_COLD]`):

`build_flags = -DNEOPIXEL_RGBW -DCOLD_WHITE -DDATA_PIN=2 ${env.build_flags} -DSECOND_SEGMENT_START_INDEX=85 -DSECOND_SEGMENT_DATA_PIN=4 -DSECOND_SEGMENT_REVERSED`

set speed to 5Mb for ESP32-S2 mini (from 2000000):
`build_flags = -DSERIALCOM_SPEED=5000000`

Probably these are the only needed changes when you recreate the fork. You dont need to delete other sections. When the Github Action finishes the build process, the final result is zip that contains all firmware versions including your modified firmware for sk6812 rgbw cold/esp32-s2 mini.

monpiere, 2023-03-05:
Thank you so much for your help! I believe that I have it setup correctly
for the most part, but I got called out just after starting the web based
configuration. So, I'll have to finish up when I get home. I will be using
a ms2109 grabber

Thanks again!
Jason

closed 2023-03-30 by awawa-dev