awawa-dev/HyperSerialESP32 #47: Relay PIN receives no current

SunnyMoo, 2023-12-13:
Hi,

I'm trying to get rid of wifi and migrate to HyperSerialESP32 from my old ESP32 with CP2102 which is flashed with WLED vanilla and drives SK9822 (GPIO 02 & 04) + power relay (GPIO12).

I got two board:
- ESP32 D1 Mini with CP2104 https://www.az-delivery.de/fr/products/esp32-d1-mini
- ESP32S with CH340C https://www.az-delivery.de/fr/products/nodemcu-esp-32s-kit
I made a fork with a branch for each board:
- ESP32D1-4000000
- ESP32S-200000
Each one has platformio.ini edited with corresponding baud rate and "LED_POWER_PIN = 12".
Compiled from github and everything went fine. Flashing with HomeFlasher is ok. They're both recognized by HyperHDR. They both output 3.3V on dedicated pin when plugged into USB...
But there's no output current on GPIO 12. Also tried with relay on GPIO 32 and it's the same for both board.

My old board has WLED configured to "invert" on power relay pin, and outputs 3.3V from GPIO 12 when plugged on USB.
I bet it's the issue with HyperSerialESP32.
I can't change that behaviour on my relay, and even if i could it would be stupid as the relay is here to power off leds when board is powered off from USB.

Thanks for you work and for your help !

awawa-dev, 2023-12-13:
Hi
WLED with "invert" disables voltage on power relay pin. Default options works the same for WLED and HyperSerialESP32 (high state if there is a communication, perform benchmark test then you will see if it works) and "invert" is simply not implemented.

SunnyMoo, 2023-12-13:
Hey awawa-dev,
I reflashed my old board with WLED 14, set relay on GPIO 12 no invert: no current and no led power ofc.
Set as inverted: 3.3V on GPIO 12 and ofc leds are powered.
So no, invert doesn't disable voltage on relay pin, it's the exact contrary. 😅

awawa-dev, 2023-12-14:
It's about control voltage that drive the power relay. When the "invert" is used then 0V (not 3.3V for WLED & HyperSerialEsp32 default relay action) is provided for power relay to "unlock" the power for LEDs. https://kno.wled.ge/features/relay-control/

SunnyMoo, 2023-12-14:
Well whatever for WLED.

I reflashed and plugged the D1:
- Relay on 3.3 pin + GPIO 12 + GND
- Data on GPIO 02
- Clock on GPIO 04
Same as old board.

Plug USB and launch HyperHDR, LED device ON: no click from the relay.

Here's the config:
[image]

awawa-dev, 2023-12-14:
Using WLED "invert" option tells us that the power relay requires LOW (0V) to activate it for the LEDs. But HyperSerialEsp32 compiled with LED_POWER_PIN provides HIGH (3.3V) on that pin. This just can't work. You can test it using multi-meter, but me and at least one other user test in when it was implemented and 3.3v was present.  Anyway that requires the communication to work properly: `perform benchmark test then you will see if it works` and provide full logs from the test but probably it doesnt work [LEDDEVICE_ADALIGHT] Device disabled, device 'adalight' signals error: 'Param�tre incorrect.'

SunnyMoo, 2023-12-14:
> Using WLED "invert" option tells us that the power relay requires LOW (0V) to activate it for the LEDs.

Well on my old board it's not that behaviour, anyway my relay is a high signal one set on opened by default.
I measured with multimeter from start and I can confirm that my relay **needs** 3.3V to close the circuit and power leds.

> HyperSerialEsp32 compiled with LED_POWER_PIN provides HIGH (3.3V) on that pin.

It should but it doesn't.

As there's a bug with the D1 I use it to provide 3.3V to relay sig and tried the ESP32S back: it works !
[image]

There's really is an issue with GPIO 12 not providing current while you tell it should.
That plus the bug with the D1.... I wonder if there's somethjing with my platformio.ini 🤔

Is this correct ?
[image]

awawa-dev, 2023-12-14:
> Is this correct ?

no it's incorrect. must have -D prefix and be in the same line as speed

SunnyMoo, 2023-12-14:
build_flags = -DSERIALCOM_SPEED=4000000 -DLED_POWER_PIN=12
This ?

awawa-dev, 2023-12-14:
I think it should work. There is a confirmation of used settings in the build log.

SunnyMoo, 2023-12-14:
Yes ! The relay is now powered by the D1, but still errors in log.
And the SP32S acts strangely: relay powers on during board boot, then turns off.

You should edit the comment header in platformio.ini to prevent similar mistakes 😅

I'm trying to debug the other issues now 😭

awawa-dev, 2023-12-14:
> And the SP32S acts strangely: relay powers on during board boot, then turns off.

Probably previous firmware didn't work at all crashing and now it works as it should when compiled properly. But here we go again: you need "inverse" ;) Or the edit code and swap/rename both powerOn()<=>powerOff() methods
https://github.com/awawa-dev/HyperSerialESP32/blob/88ed3c920d7c68d54a5c859ed83ca832953dd5c8/include/powercontrol.h#L55

> You should edit the comment header in platformio.ini to prevent similar mistakes 

Dont plan it. It's explained on the last post on my blog how to compile the firmware.

SunnyMoo, 2023-12-14:
> But here we go again: you need "inverse" ;)
No this relay needs 3.3V on signal to work.

For now I'm focusing on the D1 board which has relay powered on correctly but leds remain off.

SunnyMoo, 2023-12-14:
OK to keep orderly let's just close this issue with the solution:

build_flags = -DSERIALCOM_SPEED=4000000 -DLED_POWER_PIN=12

closed 2023-12-14 by SunnyMoo