awawa-dev/HyperSerialESP32 #52: Turning off S2 Lolin mini blue standby led

dmytro-frolov, 2024-02-12:
Hi,

Setup:
I have *ESP32-S2 Lolin mini* and *apa107/hd107* LED strip, the power goes from a 5v external power supply to buffer *SN74AHCT125N* connected to a smart socket. 

Issue:
Unfortunately, when I power off the power supply the *s2 mini* still ignites the *blue standby led* and some LEDs from the strip, I think the *SN74AHCT125N* passes some current to the strip. 

My main question is, can I turn off *s2 mini* when I don't need it, using API or something? When I connect ESP to USB, the *standby led* and LEDs don't spark until I turn on the LED device under Remote Control on the dashboard

awawa-dev, 2024-02-12:
Hi
Could you draw a simple diagram of how you connected it all? I suspect there is something wrong with the configuration (or SN74AHCT125N is damaged) because this is the first time I have heard of such a problem. If by some miracle it passes the data lines to the 5v line, would the esp32-s2 GPIO power the entire LED strip through the level shifter (this would overload the esp32 instantly)? If there is really a shortcut, +5V in the other direction also would kill the esp32 GPIO instantly.

dmytro-frolov, 2024-02-13:
Thank you for the replay,

Sure, tried to attach the diagram. Keep in mind that nothing is overloaded or overheated. If I turn off the power the ESP powers few LEDs. Those LEDs are red and dim. Until the *blue standby led* is on no matter what I do within hyperion nothing changes LEDs behaviour. If I plug out the USB and plug in back, the *standby blue led* is off as well as the LED strip.

I have 10A 5V power supply and 134 LEDs on the hd107 strip around the TV frame. 

[image]

awawa-dev, 2024-02-13:
Still this behavior is not correct and I cannot explain it from an electrical point of view how this is possible. Something's missing here. However, to me this situation looks very dangerous. Apart from the lack of power (have you checked the 5V lines with a multimeter?), the addressable LEDs will not light up without a command. Sometimes noise can trigger them, but this is rather unusual for SPI, as such an event should occur simultaneously on both SPI and DATA lines. Single-line Neopixels are much more sensitive and sometimes the fluctuations during startup (especially on GPIO2 when UART is present on esp32 or esp8266) can turn on some diodes, provided they have power.

I haven't used ESP32-s2 mini for a long time, so it may be a dead end, but I think the blue LED is connected to GPIO15 in this model... does this LED turn on again after disconnecting/connecting ESP to USB and restarting the LED driver in HyperHDR? How is the unit to which the ESP is connected powered?

awawa-dev, 2024-02-13:
> . If I turn off the power the ESP powers few LEDs. 

If you disconnect jack (LED power supply on the diagram) not esp, are they still lit?

I'm moving the subject to the discussion.