awawa-dev/HyperSerialESP32 #54: Flickering/Flashing with NodeMCU esp32 and wemos d1 mini esp8266

drbeat, 2024-03-04:
Hi,

I have a strange issue which I cannot pinpoint. Maybe you can point me to the right direction.
My Setup:
LG OLED55C7D with HyperHDR and Piccap running.
157 LEDs with external powersupply which also drives the esp32 (running WLED).

This setups works fine.

So I tried to replace it with the HyperSerial solution and HyperHDR successfully detects the esp(32 and8266 I tried both) but somehow the LEDs flash/flicker white inbetween changing color. I tried different Baudrates/Cables/ESPs but somehow the Issue persists.

But WLED works fine. I did not try HyperSerialWLED since I wanted to ditch WLED.

I am kinda confused why this happens.

awawa-dev, 2024-03-04:
Hi
If the problem concerned one board, we might suspect some incompatibility in timings with your LED strip (although we still use the Neopixelbus library just like WLED, only with slightly different methods). You don't need to change the port serial speed if it works, because packet control is maintained anyway, so we won't display the damaged one (at most we'll skip it). I suspect that something has changed in your hardware, connections or their lengths, or something has affected the ground because, for example: WLED does not have to have a direct cable connection to the unit, only to the power supply. The question is how do you power the ESP and how is everything connected now, do you have a level shifter, etc

drbeat, 2024-03-04:
Well I tried to paint my setup
[image]
this works fine without a level shifter.
[image]
and this setup does not work.

I forgot to mention that I have WS281x LEDs.

Since it did work with WLED I did not suspect that I need a level shifter.
I will try that soon und report back.

awawa-dev, 2024-03-04:
How is ESP board powered in the first (working) scenario? And is there a common ground in the second (not working) case?

awawa-dev, 2024-03-04:
>  And is there a common ground in the second (not working) case?

You need to comment ground and data line from ESP (HyperSerial) to the LED strip. I dont see it on the diagram

drbeat, 2024-03-04:
> How is ESP board powered in the first (working) scenario? And is there a common ground in the second (not working) case?

It is powered by the same PSU which also powers the LEDs. It supplies 5V to both.

> > And is there a common ground in the second (not working) case?
> 
> You need to connect ground and data line from ESP (HyperSerial) to the LED strip. I dont see it on the diagram

I did connect the data line but did not think about the ground. I will check. Thanks