awawa-dev/HyperSerialESP32 #64: Esp32c3 compatibility

Melkor82, 2024-10-26:
hello, trying to complile with the change in platformio ini for the specific board 'seeed_xiao_esp32c3' trow erroers on compilitng

`src/main.cpp:82:41: note: #pragma message: DATA_PIN = 2
 #pragma message(VAR_NAME_VALUE(DATA_PIN))
                                         ^
src/main.cpp:152:45: note: #pragma message: LED_DRIVER = NeoPixelBus<NeoGrbwFeatureNeoEsp32I2s1Sk6812Method>
  #pragma message(VAR_NAME_VALUE2(LED_DRIVER))
                                             ^
src/main.cpp:70:50: error: 'NeoEsp32I2s1Sk6812Method' was not declared in this scope
   #define LED_DRIVER NeoPixelBus<NeoGrbwFeature, NeoEsp32I2s1Sk6812Method>
                                                  ^~~~~~~~~~~~~~~~~~~~~~~~
include/base.h:46:2: note: in expansion of macro 'LED_DRIVER'
  LED_DRIVER* ledStrip1 = nullptr;
  ^~~~~~~~~~
src/main.cpp:70:50: note: suggested alternative: 'NeoEsp32Rmt1Sk6812Method'
   #define LED_DRIVER NeoPixelBus<NeoGrbwFeature, NeoEsp32I2s1Sk6812Method>
                                                  ^~~~~~~~~~~~~~~~~~~~~~~~
include/base.h:46:2: note: in expansion of macro 'LED_DRIVER'
  LED_DRIVER* ledStrip1 = nullptr;
  ^~~~~~~~~~
src/main.cpp:70:74: error: template argument 2 is invalid
   #define LED_DRIVER NeoPixelBus<NeoGrbwFeature, NeoEsp32I2s1Sk6812Method>
                                                                          ^
include/base.h:46:2: note: in expansion of macro 'LED_DRIVER'
  LED_DRIVER* ledStrip1 = nullptr;
  ^~~~~~~~~~
src/main.cpp:70:50: error: 'NeoEsp32I2s1Sk6812Method' was not declared in this scope
   #define LED_DRIVER NeoPixelBus<NeoGrbwFeature, NeoEsp32I2s1Sk6812Method>
                                                  ^~~~~~~~~~~~~~~~~~~~~~~~
include/base.h:70:10: note: in expansion of macro 'LED_DRIVER'
   inline LED_DRIVER* getLedStrip1()
          ^~~~~~~~~~
src/main.cpp:70:50: note: suggested alternative: 'NeoEsp32Rmt1Sk6812Method'
   #define LED_DRIVER NeoPixelBus<NeoGrbwFeature, NeoEsp32I2s1Sk6812Method>
                                                  ^~~~~~~~~~~~~~~~~~~~~~~~
include/base.h:70:10: note: in expansion of macro 'LED_DRIVER'
   inline LED_DRIVER* getLedStrip1()
          ^~~~~~~~~~
src/main.cpp:70:74: error: template argument 2 is invalid
   #define LED_DRIVER NeoPixelBus<NeoGrbwFeature, NeoEsp32I2s1Sk6812Method>
                                                                          ^
include/base.h:70:10: note: in expansion of macro 'LED_DRIVER'
   inline LED_DRIVER* getLedStrip1()
          ^~~~~~~~~~
include/base.h: In member function 'void Base::initLedStrip(int)':
src/main.cpp:70:50: error: 'NeoEsp32I2s1Sk6812Method' was not declared in this scope
   #define LED_DRIVER NeoPixelBus<NeoGrbwFeature, NeoEsp32I2s1Sk6812Method>
                                                  ^~~~~~~~~~~~~~~~~~~~~~~~
include/base.h:116:22: note: in expansion of macro 'LED_DRIVER'
      ledStrip1 = new LED_DRIVER(ledsNumber, DATA_PIN);
                      ^~~~~~~~~~
src/main.cpp:70:50: note: suggested alternative: 'NeoEsp32Rmt1Sk6812Method'
   #define LED_DRIVER NeoPixelBus<NeoGrbwFeature, NeoEsp32I2s1Sk6812Method>
                                                  ^~~~~~~~~~~~~~~~~~~~~~~~
include/base.h:116:22: note: in expansion of macro 'LED_DRIVER'
      ledStrip1 = new LED_DRIVER(ledsNumber, DATA_PIN);
                      ^~~~~~~~~~
src/main.cpp:70:74: error: template argument 2 is invalid
   #define LED_DRIVER NeoPixelBus<NeoGrbwFeature, NeoEsp32I2s1Sk6812Method>
                                                                          ^
include/base.h:116:22: note: in expansion of macro 'LED_DRIVER'
      ledStrip1 = new LED_DRIVER(ledsNumber, DATA_PIN);
                      ^~~~~~~~~~
In file included from include/main.h:36,
                 from src/main.cpp:173:
include/base.h:117:17: error: request for member 'Begin' in '*((Base*)this)->Base::ledStrip1', which is of non-class type 'int'
      ledStrip1->Begin();
                 ^~~~~
include/base.h: In member function 'void Base::renderLeds(bool)':
include/base.h:147:41: error: request for member 'CanShow' in '*((Base*)this)->Base::ledStrip1', which is of non-class type 'int'
     (ledStrip1 != nullptr && ledStrip1->CanShow()) &&
                                         ^~~~~~~
include/base.h:154:16: error: request for member 'Show' in '*((Base*)this)->Base::ledStrip1', which is of non-class type 'int'
     ledStrip1->Show(false);
                ^~~~
include/base.h: In member function 'bool Base::setStripPixel(uint16_t, ColorDefinition&)':
include/base.h:176:17: error: request for member 'SetPixelColor' in '*((Base*)this)->Base::ledStrip1', which is of non-class type 'int'
      ledStrip1->SetPixelColor(pix, inputColor);
                 ^~~~~~~~~~~~~
Compiling .pio\build\SK6812_RGBW_NEUTRAL\libf18\NeoPixelBus\internal\NeoPixelAnimator.cpp.o
*** [.pio\build\SK6812_RGBW_NEUTRAL\src\main.cpp.o] Error 1`

awawa-dev, 2024-10-26:
Hi
esp32-c3 does not support I2S hardware acceleration, which we use to control the LEDs in this project.

Melkor82, 2024-10-26:
ok thanks, i will order the esp32s2

awawa-dev, 2024-10-26:
You can also try latest HyperSerialWLED

awawa-dev, 2024-10-26:
version h0.14.0 should support esp32-c3..at least WLED does

Melkor82, 2024-10-26:
ok i tried that and it's working on a hyperhdr instance on a raspberry zero2w but it's not working on an instance that is running in a VM on proxmox. I see the device in the serial list but it's not working and on the vm the logs are not working too

awawa-dev, 2024-10-26:
I don't know how you made this port available in the VM, for these speeds you need direct hardware access to this port.

Melkor82, 2024-10-26:
sorry it's a container, not a vm, on which i'm running docker
the config for the porting is this:

`#<div align='center'>[image]
#
#  # Docker LXC
#
#  [image]
#  </div>
arch: amd64
cores: 4
features: nesting=1
hostname: Docker
memory: 2048
net0: name=eth0,bridge=vmbr0,gw=192.168.1.1,hwaddr=BC:24:11:3B:2C:60,ip=192.168.1.93/24,type=veth
onboot: 1
ostype: debian
rootfs: local-lvm:103/vm-103-disk-0.raw,size=10G
startup: order=2,up=30
swap: 0
tags:
lxc.cgroup2.devices.allow: a
lxc.cap.drop:
lxc.cgroup2.devices.allow: c 188:* rwm
lxc.cgroup2.devices.allow: c 189:* rwm
lxc.mount.entry: /dev/serial/by-id  dev/serial/by-id  none bind,optional,create=dir
lxc.mount.entry: /dev/ttyUSB0       dev/ttyUSB0       none bind,optional,create=file
lxc.mount.entry: /dev/ttyUSB1       dev/ttyUSB1       none bind,optional,create=file
lxc.mount.entry: /dev/ttyACM0       dev/ttyACM0       none bind,optional,create=file
lxc.mount.entry: /dev/ttyACM1       dev/ttyACM1       none bind,optional,create=file`

awawa-dev, 2024-10-26:
I'm not a docket expert but there is a risk that esp32-s2 wont work either with this config. I have esp32-c3 somewhere, will test HyperSerialWLED on C3 later

awawa-dev, 2024-10-26:
HyperSerialWLED 0.14.0 works for me on esp32-c3. Here it's receving the serial data at @ 2000000:
[image]

awawa-dev, 2024-10-26:
Make sure that device is under full container direct control, not just shared&forwared  with the host

awawa-dev, 2024-10-26:
BTW do not enable esp handshake in the Adalight for HyperSerialWLED

awawa-dev, 2024-10-27:
>  lxc.mount.entry: /dev/serial/by-id dev/serial/by-id none bind,optional,create=dir
lxc.mount.entry: /dev/ttyUSB0 dev/ttyUSB0 none bind,optional,create=file
lxc.mount.entry: /dev/ttyUSB1 dev/ttyUSB1 none bind,optional,create=file
lxc.mount.entry: /dev/ttyACM0 dev/ttyACM0 none bind,optional,create=file
lxc.mount.entry: /dev/ttyACM1 dev/ttyACM1 none bind,optional,create=file

https://forum.proxmox.com/threads/pass-usb-device-to-lxc.124205/

so probably you are wrongly sharing an already installed serial port device on the host instead of pass-through it to LXC

closed 2025-01-25 by awawa-dev