awawa-dev/HyperSerialESP32 #76: Cant find device Lg webos

accountant73, 2025-01-27:
Hi,

I am trying to setup HyperHDR on my LG 55c16A webos tv. 

I thought before soldering the led strip to test out my esp32 board with a ch340c chip. I have flashed it using esp flasher with the firmware for my led strip that. I am going to use. 

However when testing I get this error below, i have the esp32 directly using usb c connected to a usb port in the tv. 

Or should I ignore this error for now?

Logs;
2025-01-27T19:05:14.166Z [CORE] TTY is not attached to the log output
2025-01-27T19:05:14.234Z [MAIN] Database path: '/home/root/.hyperhdr/db/hyperhdr.db', readonlyMode = disabled
2025-01-27T19:05:14.234Z [MAIN] Starting HyperHdr - 20.0.0.0, (HEAD detached at v20.0.0.0) (awawa-dev-9e960c9/52f12b4-1711236142), built: Mar 28 2024:19:17:18
2025-01-27T19:05:14.234Z [MAIN] Set user data path to '/home/root/.hyperhdr'
2025-01-27T19:05:14.234Z [INSTANCE_CFG] Loading instance configuration
2025-01-27T19:05:14.244Z [DB] Database opened: /home/root/.hyperhdr/db/hyperhdr.db
2025-01-27T19:05:14.299Z [INSTANCE_CFG] Settings database initialized
2025-01-27T19:05:14.332Z [SOUND_GRABBER] Sound device is disabled
2025-01-27T19:05:14.369Z [DAEMON] The AVF grabber can not be instantiated, because it has been left out from the build
2025-01-27T19:05:14.369Z [DAEMON] The MF grabber can not be instantiated, because it has been left out from the build
2025-01-27T19:05:14.369Z [DAEMON] !The v4l2 grabber can not be instantiated, because it has been left out from the build
2025-01-27T19:05:14.380Z [DAEMON] The USB grabber was disabled during build. FlatbufferServer now controlls the HDR state.
2025-01-27T19:05:14.381Z [JSONSERVER] (JsonServer.cpp:26) Created new instance
2025-01-27T19:05:14.381Z [JSONSERVER] Started on port 19444
2025-01-27T19:05:14.383Z [SSDP] (SSDPHandler.cpp:31) SSDPHandler is initialized
2025-01-27T19:05:14.377Z [HYPERHDR0] Starting the instance
2025-01-27T19:05:14.380Z [INSTANCE_CFG0] Loading instance configuration
2025-01-27T19:05:14.381Z [DB] Database opened: /home/root/.hyperhdr/db/hyperhdr.db
2025-01-27T19:05:14.426Z [INSTANCE_CFG0] Settings database initialized
2025-01-27T19:05:14.426Z [COMPONENTCTRL0] (ComponentController.cpp:31) ComponentController is initialized. Components are ENABLED
2025-01-27T19:05:14.428Z [MUXER0] (Muxer.cpp:55) Muxer initialized
2025-01-27T19:05:14.429Z [BLACKBORDER] Set mode to: default
2025-01-27T19:05:14.430Z [IMAGETOLED_MNG0] (ImageToLedManager.cpp:192) Set LED mapping type to advanced
2025-01-27T19:05:14.430Z [IMAGETOLED_MNG0] (ImageToLedManager.cpp:176) setSparseProcessing to 0
2025-01-27T19:05:14.430Z [IMAGETOLED_MNG0] (ImageToLedManager.cpp:103) ImageToLedManager initialized
2025-01-27T19:05:14.433Z [COLORSPACE_CALIBRATION0] classicMode: no, gammas:[1.50, 1.50, 1.50], saturation: 1.00, luminance: 1.00, backLight: [enabled, threshold: 1, colored: yes]
2025-01-27T19:05:14.433Z [LED_CALIBRATION0] (LedCalibration.cpp:101) Calibration config '0' for LED segment: [0, 0]
2025-01-27T19:05:14.434Z [HYPERHDR0] Led strip RGB order is: rgb
2025-01-27T19:05:14.437Z [SMOOTHING0] High resolution clock is NOT STEADY!
2025-01-27T19:05:14.437Z [SMOOTHING0] Clearing queued colors before: enabling
2025-01-27T19:05:14.437Z [SMOOTHING0] Smoothing queue is cleared
2025-01-27T19:05:14.437Z [COMPONENTCTRL0] Smoothing: enabled
2025-01-27T19:05:14.437Z [SMOOTHING0] Creating config (0) => type: Alternative, dirMode: false, pause: false, settlingTime: 150ms, interval: 20ms (50Hz), antiFlickTres: 32, antiFlickStep: 2, antiFlickTime: 250
2025-01-27T19:05:14.437Z [SMOOTHING0] Clearing queued colors before: enabling. Smoothing configuration changed: restarting timer.
2025-01-27T19:05:14.437Z [SMOOTHING0] Smoothing queue is cleared
2025-01-27T19:05:14.437Z [SMOOTHING0] Selecting config (0) => type: Alternative, directMode: false, pause: false, settlingTime: 150ms, interval: 20ms (50Hz), antiFlickTres: 32, antiFlickStep: 2, antiFlickTime: 250
2025-01-27T19:05:14.437Z [SMOOTHING0] Selecting config (0) => type: Alternative, directMode: false, pause: false, settlingTime: 150ms, interval: 20ms (50Hz), antiFlickTres: 32, antiFlickStep: 2, antiFlickTime: 250
2025-01-27T19:05:14.438Z [LEDDEVICE0_ADALIGHT] Start LedDevice 'adalight'.
2025-01-27T19:05:14.438Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:337) deviceConfig: [{"awa_mode":true,"colorOrder":"rgb","currentLedCount":1,"espHandshake":true,"forceSerialDetection":false,"maxRetry":2,"output":"auto","rate":2000000,"refreshTime":20,"smoothingRefreshTime":20,"type":"adalight","white_channel_calibration":false}]
2025-01-27T19:05:14.438Z [LEDDEVICE0_ADALIGHT] Ignoring user LED refresh rate. Forcing smoothing refresh rate = 50.00 Hz
2025-01-27T19:05:14.438Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:423) Refresh rate = 50.00 Hz
2025-01-27T19:05:14.439Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:373) Device is not ready to start a timer
2025-01-27T19:05:14.439Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:433) Refresh interval updated to 20ms
2025-01-27T19:05:14.439Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:50) DeviceType   : adalight
2025-01-27T19:05:14.439Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:51) LedCount     : 1
2025-01-27T19:05:14.439Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:52) RefreshTime  : 20
2025-01-27T19:05:14.439Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:67) Device name   : auto
2025-01-27T19:05:14.439Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:68) Auto selection: 1
2025-01-27T19:05:14.439Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:69) Baud rate     : 2000000
2025-01-27T19:05:14.439Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:70) ESP handshake : ON
2025-01-27T19:05:14.439Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:71) Force ESP/Pico Detection : OFF
2025-01-27T19:05:14.439Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:72) Delayed open  : 0
2025-01-27T19:05:14.439Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:73) Retry limit   : 2
2025-01-27T19:05:14.439Z [LEDDEVICE0_ADALIGHT] The refresh timer is enabled ('Refresh time' > 0) and may limit the performance of the LED driver. Ignore this error if you set it on purpose for some reason (but you almost never need it).
2025-01-27T19:05:14.439Z [LEDDEVICE0_ADALIGHT] (LedDeviceAdalight.cpp:79) Adalight driver with activated high speeed & data integration check AWA protocol
2025-01-27T19:05:14.439Z [LEDDEVICE0_ADALIGHT] (LedDeviceAdalight.cpp:88) Adalight header for 1 leds: Awa 0x00 0x00 0x55
2025-01-27T19:05:14.439Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:247) Enable the device
2025-01-27T19:05:14.443Z [HYPER_MANAGER] HyperHDR instance 'First LED instance' has been started
2025-01-27T19:05:14.443Z [HYPERHDR0] The instance is running
[... 113 lines cut ...]
2025-01-27T19:05:19.251Z [SMOOTHING0] Selecting config (0) => type: Alternative, directMode: false, pause: false, settlingTime: 150ms, interval: 20ms (50Hz), antiFlickTres: 32, antiFlickStep: 2, antiFlickTime: 250
2025-01-27T19:05:19.251Z [SMOOTHING0] Using alternative smoothing input (0)
2025-01-27T19:05:19.251Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:381) Stopping refresh timer
2025-01-27T19:05:19.251Z [HYPERHDR0] New priority[255], previous [0]
2025-01-27T19:05:19.251Z [HYPERHDR0] No source left -> switch LED-Device off
2025-01-27T19:05:19.251Z [SMOOTHING0] Clearing queued colors before: disabling
2025-01-27T19:05:19.251Z [LEDDEVICE0_ADALIGHT] Ignoring user LED refresh rate. Forcing smoothing refresh rate = 50.00 Hz
2025-01-27T19:05:19.251Z [SMOOTHING0] Smoothing queue is cleared
2025-01-27T19:05:19.251Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:423) Refresh rate = 50.00 Hz
2025-01-27T19:05:19.251Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:367) Starting timer with interval = 20ms
2025-01-27T19:05:19.251Z [COLORSPACE_CALIBRATION0] (ColorSpaceCalibration.cpp:225) setBackLightEnabled: 0
2025-01-27T19:05:19.251Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:433) Refresh interval updated to 20ms
2025-01-27T19:05:19.251Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:448) LED refresh interval adjustment caused by smoothing configuration change to 20ms (proposed: 20ms)
2025-01-27T19:05:19.251Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:297) Disable the device
2025-01-27T19:05:19.251Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:381) Stopping refresh timer
2025-01-27T19:05:19.251Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:90) Switch off
2025-01-27T19:05:19.251Z [LEDDEVICE0_ADALIGHT] (LedDevice.cpp:544) Set LED strip to black/power off
2025-01-27T19:05:19.252Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:146) Flush was successful
[... cut ...]
2025-01-27T19:05:19.253Z [EFFECTENGINE0] Effect 'Rainbow swirl fast' has finished.
2025-01-27T19:05:19.253Z [EFFECT0(Rainbo...)] Effect named: 'Rainbow swirl fast' is deleted
2025-01-27T19:05:19.853Z [LEDDEVICE0_ADALIGHT] (ProviderSerial.cpp:164) Serial port is closed: ttyAMA0
2025-01-27T19:05:19.853Z [COMPONENTCTRL0] LED device: disabled
2025-01-27T19:05:27.206Z [WEBSOCKET] (WebSocketClient.cpp:39) New connection from ::ffff:192.168.178.76
2025-01-27T19:05:27.832Z [WEBSOCKET] (HyperAPI.cpp:957) log streaming activated for client ::ffff:192.168.178.76

awawa-dev, 2025-01-27:
Hi
For ch340c on webos you need kernel driver for this uart. Or save yourself the trouble of tinkering with the OS and replace the esp32 with a regular Pico/rp2040. Or ideally with an Adafruit/Pimoroni board with a built-in level shifter.

awawa-dev, 2025-01-27:
https://github.com/satgit62/How-to-Install-and-set-up-Ambilight-on-LG-webOS?tab=readme-ov-file#other-led-controllers

accountant73, 2025-01-28:
Thanks, I have bought this one https://www.robotshop.com/products/pimoroni-plasma-2040-neopixel-led-strip-controller