awawa-dev/HyperSerialESP32 #82: Wiring esp32 + 12v ws2815 without blowing the esp

kekkokk, 2025-03-02:
Hi guys!

I'm trying to drive 309 WS2815 12V LEDs using HyperSerialESP32 and a 12v 8.5A PSU.

I have HyperHDR installed on my LG C1 TV with the required CP2104 kernel driver.

My first attempt was simply connecting GPIO2 to the data line of the LED strip, but it resulted in a flickering mess.
So, I grounded the 12V power supply to the ground pin of the ESP32, and magically, I got an almost perfect result though I still noticed some flickering.

I went into HyperHDR and started messing around with colors: it seems that it doesn't like bright colors, such as cyan. It flickers a lot.

Also with both gorund connected. if I turn off the power to the esp32 (disconnect usb) and reconnect, it shows a rainbow static color but does not work unless I disconnect and reconnect the wire that ground esp <---> led strip (this is strange behaviour, but consistent)

\> **Adding Level shifter**
I then moved on to the setup with the I²C level shifter, but I really don't understand how to wire it properly withour burning the esp.
I tried this solution (which is exactly the same setup I'm using to drive the same type of LEDs on my PC monitor, where I have half the number of pixels, and it works like a charm).

Here’s the schematic (showing only one strip, and the I²C converter is wired correctly, not as shown in the diagram where the LV and HV pins are inverted):
[Schematic](https://global.discourse-cdn.com/free1/uploads/wled/original/2X/6/68fcaa161d04ec8f6508f3e9c5a2a3180980ed46.jpeg)

I know this level shifter is not the optimal one, as it might be too slow, but it has been working perfectly in my other setup for over a year. While waiting for the correct level shifter to arrive, I decided to run some tests.

However, somehow my ESP32 broke after a short period of time, and now the USB is no longer recognized. This is the second time it has happened, so I must be doing something very wrong which causes my esp board to blow up.

Everywhere online, I read that you should not simultaneously connect the USB while powering the board with an external 5V supply.
But then, I don't understand how I should wire the level shifter since I need the ESP32 to be connected to the USB serial on my TV while also providing it with 5V on VCC.

Can you please explain how I'm supposed to connect everyting or possibly draw an sketch of how to properly wire an ESP32 to drive a 12V LED strip?
I feel so dumb right now, and I really want to avoid burning a third ESP32.

Thanks a lot for your effort into the project

awawa-dev, 2025-03-02:
Hi 12v to 5v converter is completely unnecessary. For the level shifter you can take 5v from the esp32  board.

kekkokk, 2025-03-02:
So only using usb power? And should I connect the ground to the 12v I suppose.

Is there any prebuilt solution?

awawa-dev, 2025-03-02:
Level shifter ground and data output to the led strip. Alternatively you can use one if Pico board with built-in level shifter. Rp2040 doesn't need special drivers on LG  webos. Check out the HyperSerialPico project for more details

kekkokk, 2025-03-04:
Went for an Adafruit ItsyBitsy RP2040 and works like a charm.

Thanlks a lot!

closed 2025-03-04 by kekkokk